warwick320/nova-x #3: There is no effect when attacking.

7JKD, 2024-10-24:
It cres when attacking wifi, and it has no effect. What's going on?

samukas81, 2024-10-24:
Can help me?

C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp: In function 'void wifi_tx_raw_frame(void*, size_t)':
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp:12:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
   void *frame_control = alloc_mgtxmitframe(ptr + 0xae0);
                                                  ^~~~~
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp:15:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     update_mgntframe_attrib(ptr, frame_control + 8);
                                                  ^
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp:16:50: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     memset((void *)*(uint32_t *)(frame_control + 0x80), 0, 0x68);
                                                  ^~~~
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp:17:68: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     uint8_t *frame_data = (uint8_t *)*(uint32_t *)(frame_control + 0x80) + 0x28;
                                                                    ^~~~
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp:19:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     *(uint32_t *)(frame_control + 0x14) = length;
                                   ^~~~
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\wifi_cust_tx.cpp:20:35: warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
     *(uint32_t *)(frame_control + 0x18) = length;
                                   ^~~~
In file included from c:\Users\Dell\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:13:0,
                 from C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:14:
c:\Users\Dell\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:95:41: error: 'SPI_MODE0' was not declared in this scope
                      uint8_t dataMode = SPI_MODE0, SPIClass *theSPI = &SPI);
                                         ^~~~~~~~~
c:\Users\Dell\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:104:41: error: 'SPI_MODE0' was not declared in this scope
                      uint8_t dataMode = SPI_MODE0);
                                         ^~~~~~~~~
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\RTL8720dn-Deauther.ino: In function 'void loop()':
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:294:13: warning: unused variable 'deauth_reason2' [-Wunused-variable]
         int deauth_reason2 = 1;
             ^~~~~~~~~~~~~~
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\RTL8720dn-Deauther.ino: In function 'String getMenuName(int)':
C:\Users\Dell\Downloads\5G-deauther-with-OLED-SSD1306-main\5G-deauther-with-OLED-SSD1306-main\RTL8720dn-Deauther\RTL8720dn-Deauther.ino:324:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

exit status 1

Compilation error: exit status 1

WXLN, 2024-11-05:
Like me, the attack is also ineffective
[image]
[image]
Although it looks beautiful, it doesn't work and makes people feel helpless

samukas81, 2024-11-05:
The problem is that the person copies a code without knowing if it works.

closed 2024-12-23 by warwick320