warwick320/nova-x #13: error: 'SPI_MODE0' was not declared in this scope

jerry1233, 2025-01-01:
error: 'SPI_MODE0' was not declared in this scope

Environment:

Adafruit_BusIO 1.10.0
Adafruit-GFX-Library 1.11.11
Adafruit_SSD1306 2.5.13
arduino ide 2.3.4

Error:

In file included from c:\Users\admin\Documents\Arduino\libraries\Adafruit_GFX_Library/Adafruit_GFX.h:13:0,
                 from C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:17:
c:\Users\admin\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:70:41: error: 'SPI_MODE0' was not declared in this scope
                      uint8_t dataMode = SPI_MODE0, SPIClass *theSPI = &SPI);
                                         ^~~~~~~~~
c:\Users\admin\Documents\Arduino\libraries\Adafruit_BusIO/Adafruit_SPIDevice.h:75:41: error: 'SPI_MODE0' was not declared in this scope
                      uint8_t dataMode = SPI_MODE0);
                                         ^~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void drawssid()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:187:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       if(scrollindex < scan_results.size() - 1){
          ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:223:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(int i=0;i<SelectedVector.size();i++){
                 ~^~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void Multi()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:276:29: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
       if(digitalRead(BTN_OK)==LOW | digitalRead(BTN_BACK)==LOW){
                              
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:284:14: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       if(num >= SelectedVector.size()){
          ~~~~^~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void Single()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:305:27: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
     if(digitalRead(BTN_OK)==LOW | digitalRead(BTN_BACK)==LOW){
                            
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void All()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:329:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(int i = 0; i<scan_results.size();i++){
                    ~^~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:330:29: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
       if(digitalRead(BTN_OK)==LOW | digitalRead(BTN_BACK)==LOW){
                              
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void BecaonDeauth()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:360:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(int i = 0; i<scan_results.size();i++){
                    ~^~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:361:29: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
       if(digitalRead(BTN_OK)==LOW | digitalRead(BTN_OK)==LOW){
                              
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void Becaon()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:390:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for(int i = 0; i<scan_results.size();i++){
                    ~^~~~~~~~~~~~~~~~~~~~
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:391:29: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
       if(digitalRead(BTN_OK)==LOW | digitalRead(BTN_BACK)==LOW){
                              
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino: In function 'void RandomBeacon()':
C:\Users\admin\Downloads\OLED3.5\OLED3\OLED3.ino:428:27: warning: suggest parentheses around comparison in operand of '|' [-Wparentheses]
     if(digitalRead(BTN_OK)==LOW | digitalRead(BTN_BACK)==LOW){
                            

exit status 1

Compilation error: exit status 1

warwick320, 2025-01-02:
Use an image upload tool

jerry1233, 2025-01-09:
> Use an image upload tool
I want try figure out why and use different gpio point😁