warwick320/nova-x #19: Errors in uploading on Arduino IDE

leosel, 2025-05-03:
In file included from c:\users\leo\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\stl_tree.h:63:0,
                 from c:\users\leo\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\map:60,
                 from C:\Users\Leo\Documents\Arduino\BW16\BW16.ino:5:
c:\users\leo\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\stl_algobase.h:243:56: error: macro "min" passed 3 arguments, but takes just 2
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
c:\users\leo\appdata\local\arduino15\packages\realtek\tools\ameba_d_asdk_toolchain\1.0.1\arm-none-eabi\include\c++\6.5.0\bits\stl_algobase.h:265:56: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
                                                        ^
exit status 1

Compilation error: exit status 1

warwick320, 2025-05-03:
[image]

warwick320, 2025-05-03:
You need to enable Arduino_STD_PRINTF

closed 2025-05-10 by warwick320