warwick320/nova-x #27: Hi ， are u considering writing a Chinese UI for this  SSD1306 OLED  ？ Please

XMG114514, 2025-05-30:
ENGLISH UI is unhelp for Chinese

XMG114514, 2025-05-31:
wow , thank u bro. I'm looking forward to your screen animation. If you encounter any problems with Chinese when considering updating U8g2, maybe I can help you

XMG114514, 2025-06-02:
Emm.If you feel it's too big or not suitable for you, you can send me the code you tried and I will try to optimize it. If successful, I will send it back to you. I believe there will be many people who like it. Let's solve the problem together!

warwick320, 2025-06-03:
It's easier than I expected. However, there's one issue: when switching to the U8g2 library, I run into a problem with Realtek Board Manager (3.1.8). To resolve this, I need to downgrade the Board Manager, so I don't think I'll be able to use the WebUI function.

XMG114514, 2025-06-03:
Well, I don't know much about the code. Are you saying that the screen can already display Chinese? But can't the web version display Chinese?

XMG114514, 2025-06-03:
Can you send me the code for successfully using u8g2? I couldn't find it in your open-source project. Maybe we can add an email communication method? This way of communication will be much more convenient

warwick320, 2025-06-03:
No. In Realtek board manager 3.1.8, if I use Wire.begin(); or u8g2.begin();, I get errors and the code doesn't work. I don't know why, but if I downgrade the version to 3.1.5, I can use it. However, in my case with 3.1.5, when I try to connect to the BW16 board AP, there's a problem where I can't get the IP address. So if I downgrade to 3.1.5, I can't use the webui function.

XMG114514, 2025-06-03:
The only problem now is the display of Chinese characters on the screen? Has everything in the backend been resolved? So it seems like you're really great, qwq

XMG114514, 2025-06-03:
你方便接受图片吗？我这里有个使用u8g2的项目 可以正常显示汉字 你或许可以看看？

warwick320, 2025-06-03:
Displaying Chinese characters is easy, but the character width and roundrect width are different from the original, so I think I just need to fix that.

XMG114514, 2025-06-03:
good 
Can you update the open source project recently or are you free this weekend?

XMG114514, 2025-06-03:
Okay, do you want a brief translation or a complete translation? For example, a short translation of 4 words or a complete translation of 2 words may be too long to fit on a single line

warwick320, 2025-06-03:
I'd like a brief translation

XMG114514, 2025-06-03:
Deauthentication Flood  认证攻击
Beacon                           信标攻击
Association Flood           洪水攻击
SSID Cloning                  SSID克隆

XMG114514, 2025-06-03:
Simplify things into the first two words  and the last two words mean attack  If you think it's inappropriate  I can also change his meaning into four words  But you can't attack later Otherwise the screen may not fit

XMG114514, 2025-06-03:
I have encountered some problems. Are the first two you mentioned, beacon and ssid cloning, sending attack frames or creating a large number of fake wifi? If it's the latter, isn't ssid cloning also cloning wifi?

XMG114514, 2025-06-03:
Deauthentication Flood  认证攻击
Beacon Flood   随机AP
Association Flood  关联洪水
Authentication Flood   洪水攻击
SSID Cloning     SSID克隆

XMG114514, 2025-06-03:
I still think these Chinese names are a bit incomplete. Of course, that's okay. If you want more easily understandable and complete Chinese characters, could you explain in detail the specific attack methods for each item? The Chinese characters I gave you above were just summarized based on their literal meaning and my understanding

XMG114514, 2025-06-04:
It seems that the following meanings of chat_gpt translation are not very correct and easy to understand

XMG114514, 2025-06-04:
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_3wire_hw_spi(u8x8_t*, uint8_t, uint8_t, void*)':
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:819:45: error: 'SPI_MODE0' was not declared in this scope
                 case 0: internal_spi_mode = SPI_MODE0; break;
                                             ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:820:45: error: 'SPI_MODE1' was not declared in this scope
                 case 1: internal_spi_mode = SPI_MODE1; break;
                                             ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:821:45: error: 'SPI_MODE2' was not declared in this scope
                 case 2: internal_spi_mode = SPI_MODE2; break;
                                             ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:822:45: error: 'SPI_MODE3' was not declared in this scope
                 case 3: internal_spi_mode = SPI_MODE3; break;
                                             ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_hw_spi(u8x8_t*, uint8_t, uint8_t, void*)':
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:964:30: error: 'SPI_MODE0' was not declared in this scope
  case 0: internal_spi_mode = SPI_MODE0; break;
                              ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:965:30: error: 'SPI_MODE1' was not declared in this scope
  case 1: internal_spi_mode = SPI_MODE1; break;
                              ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:966:30: error: 'SPI_MODE2' was not declared in this scope
  case 2: internal_spi_mode = SPI_MODE2; break;
                              ^~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:967:30: error: 'SPI_MODE3' was not declared in this scope
  case 3: internal_spi_mode = SPI_MODE3; break;
                              ^~~~~~~~~
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\Administrator\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi
  Not used: C:\Users\Administrator\Documents\Arduino\libraries\WiFiEspAT
Using library AmebaWire at version 1.0.0 in folder: C:\Users\Administrator\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\Wire 
Using library U8g2 at version 2.35.30 in folder: C:\Users\Administrator\Documents\Arduino\libraries\U8g2 
Using library WiFi at version 1.0.0 in folder: C:\Users\Administrator\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\WiFi 
Using library AmebaSPI at version 1.0.0 in folder: C:\Users\Administrator\AppData\Local\Arduino15\packages\realtek\hardware\AmebaD\3.1.7\libraries\SPI 
exit status 1

Compilation error: exit status 1

XMG114514, 2025-06-04:
It seems that I have encountered some problems? Perhaps it's an issue with my storage environment?

warwick320, 2025-06-04:
Did you check Standard Lib?

warwick320, 2025-06-04:
Try [this](https://github.com/warwick320/NovaX-5G-RTL8720dn-Bw16-Deauther/releases/tag/ImageTool_v2C). It can upload already compiled bin file.

XMG114514, 2025-06-04:
Ok  thank you

warwick320, 2025-06-04:
Oh, you translated it. I just saw it. Thank you.

XMG114514, 2025-06-04:
[image]
[image]
[image]

warwick320, 2025-06-04:
> c:\Users\Administrator\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_3wire_hw_spi(u8x8_t*, uint8_t, uint8_t, void*)': [...]

add this #include <SPI.h> before #include <U8g2lib.h>in display.h

XMG114514, 2025-06-04:
Of course

XMG114514, 2025-06-04:
ok

XMG114514, 2025-06-04:
per deauth frame
一次认证帧数
per beacon frame
一次信标帧数
Max Clone
Max克隆数量
4.Max Spam
Max干扰数量

XMG114514, 2025-06-04:
Try this

XMG114514, 2025-06-04:
per deauth frame
认证帧/次
per beacon frame
信标帧/次
Max Clone
Max克隆数
4.Max Spam
Max干扰数

XMG114514, 2025-06-04:
for  “max”  You don't have to translate  , Most people can understand it

XMG114514, 2025-06-04:
If you want to translate max   it it 最大

XMG114514, 2025-06-04:
If you want to translate max it is  最大

WXLN, 2025-06-04:
> [image] [image] [image]

这个其实就是warwick320做的。。只是国内的某眼做了汉化。，

XMG114514, 2025-06-04:
我知道啊 因为相关代码都非常像只不过我不会汉化 极客也没开源而已。。。。

XMG114514, 2025-06-04:
不是 那是因为他跟不你不想放多选版本的！ 你没看到他的论坛相关文件什么都撤销了  他只是怕出事  他手里绝对有多选的  但是他不可能放出来

XMG114514, 2025-06-04:
我不这么认为  你应该在他的立场下考虑  确实是开源的项目没有错  但是他如果发布了多选对更多人 造成了更加大的危害等等  那他可能会面临起诉 等问题 我觉得他放出中文版的已经很好了  也是弥补了没有中文带屏的固件的缺点  虽然是单选的版本吧

XMG114514, 2025-06-10:
emm 中文版的只有屏幕可以使用 而web ui 是无法使用的？只有英文版本的可以吗？