warwick320/nova-x #30: 网页功能不稳定

WXLN, 2025-06-04:
You can only connect to web pages occasionally. The main problem is that after connecting to WIFI, it will automatically disconnect when entering 192.168.1.1. dysfunction

WXLN, 2025-06-05:
> 我没有发现任何问题。请问您是否在启用 Web UI 功能的情况下访问了它（无论是在首次启动时，还是在初始屏幕上按返回按钮）？

yes

https://github.com/user-attachments/assets/ddc0dd0d-74a7-4f31-bec6-b61574922af5

warwick320, 2025-06-06:
There is no problem on my phone. Does [this ](https://github.com/tesa-klebeband/RTL8720dn-Deauther)work?

warwick320, 2025-06-07:
Is your board manager version 3.1.8?

WXLN, 2025-06-09:
> 你的Board Manager是3.1.8版本吗？
The reason has been identified, it is a power supply issue. It's related to the screen with buttons that I use. Thank you for your help

closed 2025-06-09 by warwick320