warwick320/nova-x #31: Bro, can you make a version with antennas for the ESP32 and BW16?

Batmanmz, 2025-06-09:
Bro, can you make a version with antennas for the ESP32 and BW16?

XMG114514, 2025-06-10:
are you ok ?硬件问题你让作者写软件来实现？？你为什么不买带ipex版本的bw16 ？or自己改电路让他支持ipex天线不就好了

WXLN, 2025-06-19:
> 兄弟，你能制作一个带天线的 ESP32 和 BW16 版本吗？

This is a hardware issue and has nothing to do with the author's source code.You need to design your own circuit, as shown in the diagram

[image]
[image]

warwick320, 2025-06-21:
That is really cool

XMG114514, 2025-06-21:
Do you know where the animation speed is in the source code? I feel like the movement speed of the box is a bit slow

XMG114514, 2025-06-21:
yes

warwick320, 2025-06-21:
Reduce animation frame

warwick320, 2025-06-21:
line 144

warwick320, 2025-06-21:
> > 兄弟，你能制作一个带天线的 ESP32 和 BW16 版本吗？
> 
> This is a hardware issue and has nothing to do with the author's source code.You need to design your own circuit, as shown in the diagram
> 
> [image] [image]

Where did you get the schematic for the BW16 board?

warwick320, 2025-06-21:
Ow Thanks

warwick320, 2025-06-25:
> > > > 兄弟，你能制作一个带ESP32和BW16版本的天线吗？
> > > 
> > > 
> > > 这是硬件问题，与作者的源代码无关。您需要自己设计电路，如图所示
> > > [image] ![图像](https://private-user-images.githubusercontent.com/127818672/456813789-2eda61b2-ab09-42c3-a908-7a1ec6e8f1cf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTA2NzgwMzksIm5iZiI6MTc1MDY3NzczOSwicGF0aCI6Ii8xMjc4MTg2NzIvNDU2ODEzNzg5LTJlZGE2MWIyLWFiMDktNDJjMy1hOTA4LTdhMWVjNmU4ZjFjZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDYyM1QxMTIyMTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYmZmZThhNjAwMzUyZGZjYmU5OTQ4YTg4NjFhNzNlZWVmYTdlY2JiZjVkNjc4MTI4MTg5ZmQwMzdhMzA1ODgyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.KtfUTjh61Na4DQH4KlsbaajgVgusXMzwwMfQ445lgw0)
> > 
> > 
> > 您从哪里获得 BW16 板的原理图？
> 
> [bw16-kit_v1.2.3_规格书-sch-specification_cn.pdf](https://github.com/user-attachments/files/20864294/bw16-kit_v1.2.3_.-sch-specification_cn.pdf)
> 
> I designed it based on the official schematic provided. Removed some unnecessary circuits.

[image]
In this BW16 schematic, how did you handle the NC resistors and the 0R resistors? Can I just leave the NC resistors unconnected and only connect the 0R resistors?
I don't know much about PCBs.

WXLN, 2025-06-26:
> > > > > 兄弟，你能制作一个带ESP32和BW16版本的天线吗？
> > > > 
> > > > 
> > > > 这是硬件问题，与作者的来源代码相关。您需要自己设计电路，如图所示
> > > > [image] ![图像](https://private-user-images.githubusercontent.com/127818672/456813789-2eda61b2-ab09-42c3-a908-7a1ec6e8f1cf.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTA2NzgwMzksIm5iZiI6MTc1MDY3NzczOSwicGF0aCI6Ii8xMjc4MTg2NzIvNDU2ODEzNzg5LTJlZGE2MWIyLWFiMDktNDJjMy1hOTA4LTdhMWVjNmU4ZjFjZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDYyM1QxMTIyMTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYmZmZThhNjAwMzUyZGZjYmU5OTQ4YTg4NjFhNzNlZWVmYTdlY2JiZjVkNjc4MTI4MTg5ZmQwMzdhMzA1ODgyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.KtfUTjh61Na4DQH4KlsbaajgVgusXMzwwMfQ445lgw0)
> > > 
> > > 
> > > 您从哪里获得BW16板的原理图？
> > 
> > 
> > [bw16-kit_v1.2.3_规格书-sch-specation_cn.pdf](https://github.com/user-attachments/files/20864294/bw16-kit_v1.2.3_.-sch-specification_cn.pdf)
> > 我根据官方提供的原理图设计了它。删除了一些不必要的电路。
> 
> [image] 在这个 BW16 原理图中，你是如何处理常闭电阻和零电阻的？我可以不连接常闭电阻，只连接零电阻吗？ 我对 PCB 不太了解。

In some cases, NC needs to be short circuited. So it needs to be preserved. For example, when burning。It is best to follow its schematic directly, or based on  https://oshwhub.com/9527iii/bw16-sha-shou-5gmini-ban
You can refer to the information in this link, but the production of his PCB is really poor.But if you are not proficient in PCB production, you can tell me your email and I will send you a PCB file that I have made

warwick320, 2025-06-26:
If you could kindly share the PCB file, I would be very grateful and make good use of it. Thank you so much in advance. My email address is sihu320@gmail.com.

WXLN, 2025-06-26:
> 如果您能分享一下PCB文件，我将不胜感激，并会好好利用它。在此先行致谢。我的邮箱地址是[sihu320@gmail.com](mailto:sihu320@gmail.com)。

Okay, if you have your own exclusive logo (image), you can. Add to the project。Then we will create a PCB with your exclusive logo for you.It is best to use PNG/JPG format.

XMG114514, 2025-06-27:
> In some cases, NC needs to be short circuited. So it needs to be preserved. For example, when burning。It is best to follow its schematic directly, or based on [...]

In fact, BW16 only requires a 3.3V power supply to work and does not require any external pull-up or pull-down resistors. However, for burning, it needs to be used. You can first burn it on the burning rack and then place it on the PCB. That's how I did it

WXLN, 2025-06-28:
> 太感谢了

[image]
[image]
Sent to your email, please check your inbox.

warwick320, 2025-06-29:
Wow that’s really cool

damak07, 2025-07-02:
if someone sell this pcb or ready unit i am interested