warwick320/nova-x #33: Please create a .bin file with the new updates — the code doesn't work through the Arduino IDE, it just freezes.

Batmanmz, 2025-06-12:
Please create a .bin file with the new updates — the code doesn't work through the Arduino IDE, it just freezes. And if possible, try to implement data transmission between the BW16 and ESP32 via RX and TX — maybe this will solve the issue with Evil Twin and Apple.

warwick320, 2025-06-13:
I want to make ESP32 and Bw16 communicate with TX RX, but I'm a Korean student and I don't have time because of the exam period. I'll upload the bin file now.

warwick320, 2025-06-13:
And if you don't know how to upload a bin file, try using the Image Upload tool. [Link](https://github.com/warwick320/NovaX-5G-RTL8720dn-Bw16-Deauther/releases/tag/ImageTool_v2)

closed 2025-06-13 by warwick320

Batmanmz, 2025-06-13:
All the recent .bin files I installed work, but the .ino ones don’t. Thanks for the updated .bin—I tried making a .bin from the latest .ino, but it also freezes during an attack. Your .bin files work fine, and I’d like to change the SSID name and password to my own if possible. Hope you do well on your exams!!!